ElectronicCats/ElectronicCats-PN7150 #16: I2C in ESP32

hkarthik97, 2020-08-12:
I tried to connect ESP32 and PN7150 plug and play device. I connected in I2C. The serial monitor shows Setup Ok. But when I scan the card nothing happens. Which microcontroller did you use?

sabas1080, 2020-08-12:
Hi @hkarthik97 

This library is not ready to work in ESP32 yet, apparently there is not enough buffer for the responses of the PN7150, but welcome your pullrequest, check wire library

https://github.com/ElectronicCats/ElectronicCats-PN7150/blob/master/src/Electroniccats_PN7150.h#L23-L32

this library has been tested on SAMD21, but should work on SMT32, Teensy and similary boards

sabas1080, 2020-10-08:
Hi @hkarthik97 this library work for you in ESP32?

hkarthik97, 2020-10-08:
Yes it works for me in esp32

closed 2020-10-08 by sabas1080

HighCodeh, 2024-09-16:
> Sim, funciona para mim no esp32

Hello, I wanted to know how you implemented it for use on the es32? I'm trying to use the pn7150 with the esp32 but I'm not succeeding!!

Eric286, 2024-09-17:
Hello @HighCodeh 

Please verify with an I2C scanner that the module is recognized to eliminate the possibility that this is the problem.

I also recommend trying with another card, if you have it available, to see if the error is directly due to the ESP32 or not.

Kind Regards!

HighCodeh, 2024-09-21:
Could you tell me which pins are used in the respective pins of the pn7150 module?