ElectronicCats/ElectronicCats-PN7150 #17: Is it possible to read the data stored in the card

hkarthik97, 2020-09-17:
I am storing some text in my card. I am not able to read the data will this library read the data stored in the card?
Thanks in advance

salmg, 2020-09-21:
What type of card? What type of technology? Which hardware did you use to write it? Which example are you using to read it? Do you have some code?.....

Difficult to guess all those variables.

hkarthik97, 2020-09-24:
I  want to transfer text from my phone NFC via NDEF.I am using PN7150. I am used both Detect tags and Readblock mifare classic example

hkarthik97, 2020-10-08:
Any update on this?

hkarthik97, 2020-10-14:
Please help .. 
Will don/ndef library help?

sabas1080, 2020-10-14:
@hkarthik97 NDEF it is a feature we are working on

hkarthik97, 2020-10-14:
> @hkarthik97 NDEF it is a feature we are working on

When is it expected to be released?

salmg, 2020-10-14:
Soon

hkarthik97, 2020-11-24:
When will you release the update for P2P?

sabas1080, 2020-11-26:
@hkarthik97 P2P work in beta, yo can clone this repo, but NDEF no yet, we are working

closed 2022-10-08 by stale[bot]

sabas1080, 2023-07-29:
#25