ElectronicCats/ElectronicCats-PN7150 #19: NxpNci_TVDD_CONF_2ndGen doesn't appear to reflect User Manual

JWizlo, 2020-10-12:
Hello, looking at the User Manual it says Byte 2 of PCM_CFG should be 0x00 for CFG1 and 0x01 for CFG2.

I see in this code 0x066400 is written but I'm curious if it should instead be 0x066401.

The data in question is stored in NxpNci_TVDD_CONF[] in Electroniccats_PN7150.cpp line 716.

I can't test this myself at the moment so I wanted to reach out and see if this discrepancy was known.

salmg, 2020-10-14:
at this point we are testing the nxp prototype with this configuration which we ported from linux driver, and it seems that it is working properly; but we will continue testing to see any discrepancies in the hardware side, thanks

JWizlo, 2020-10-14:
Okay. I was just curious if you had another datasheet that described the RFU registers. If the data came from the driver then that is probably the end of that line of inquiry. Thanks!

closed 2020-10-26 by salmg