ElectronicCats/ElectronicCats-PN7150 #22: Is there a way to use this library in a nonblocking way?

schmidty1913, 2021-01-24:
nfc.WaitForDiscoveryNotification(&RfInterface) is a blocking function, is there a way to have it non blocking?

salmg, 2021-01-25:
If you explain what you want to do, we can see if there is other way to do it. That function specifically is designed to be like that because is always looking for cards

schmidty1913, 2021-01-25:
I am looking to have the library wait for a card but also be able to loop.

I am looking to receive serial commands and also blink an LED.

salmg, 2021-01-25:
what we can do is to add a parameter to the function to make it more flexible about timeout; i will add this change to the next modification, thanks

schmidty1913, 2021-02-06:
I was thinking it would be better to have a function that checks for the IRQ pin. Call that function and it returns false if the IRQ is not set and returns card data if the IRQ is set

salmg, 2021-03-07:
yes please, add the functionality, and we can review it

sabas1080, 2021-03-28:
@salmg @schmidty1913 

I have pullrequest with interrupt and IRQ #24

sabas1080, 2022-11-14:
Now in #46

closed 2022-11-14 by sabas1080