ElectronicCats/ElectronicCats-PN7150 #28: Issue with OM5578/PN7150ARDM

m1ckey, 2021-09-06:
I would like to use this library, but cannot get the example "DetectTags" to work.

My hardware is OM5578/PN7150ARDM and Arduino Mega R3 2560.
I modified in the DetectTags.ino the pin definitions (IRQ=8, VEN=7).
If I start the program it fails in `Electroniccats_PN7150::ConfigureSettings`:
https://github.com/ElectronicCats/ElectronicCats-PN7150/blob/b0297f30d0117d373a0399a2c9f43245de281490/src/Electroniccats_PN7150.cpp#L1274
(The rxBuffer is all zeros)

If I simply skip Electroniccats_PN7150::ConfigureSettings in the example code, the PN7150 behaves very weird.
(It successfully reads the first card, but cannot read another card)

I am not sure how to debug this further and would appreciate any input.

Michael

PS:
The output from Electroniccats_PN7150::connectNCI:
```
0xROM_CODE_V: 10
FW_MAJOR_NO: 1
0xFW_MINOR_NO: a0
gNfcController_generation: 2
```

wilson-elechouse, 2022-02-17:
I tested the code with Arduino Miro. I have the same error:
0xROM_CODE_V: 10
FW_MAJOR_NO: 1
0xFW_MINOR_NO: a0
gNfcController_generation: 2

m1ckey, 2022-06-23:
Thank you for reaching out.  
Sorry, I do not have any updates. I ended up using other hardware.

jazhe, 2022-06-23:
Hello @m1ckey! Thanks for the information, would you like to receive an update on our findings? If not, please let us know if we can close this issue now.
Also, thanks for reporting this issue, feedback like this helps us improve!

> Thank you for reaching out. Sorry, I do not have any updates. I ended up using other hardware.

m1ckey, 2022-07-02:
> If not, please let us know if we can close this issue now.

closed 2022-07-02 by m1ckey

wilson-elechouse, 2022-10-11:
Hi,
I would like to receive the updating.
We are still waiting for the problem to be resolved.
Thanks

jazhe, 2022-10-12:
Hello @wilson-elechouse!

Thanks for your patience.
Unfortunately, the issue will persist due to incompatibility with AVR boards (Uno, Mega, Micro, etc), the library was not created for 8-bit microcontrollers. 
Apologies for the inconvenience caused.

Hope you have a nice day!

wilson-elechouse, 2024-06-13:
Will this library supports PN7160?

jazhe, 2024-06-14:
Hello @wilson-elechouse, we have not been able to validate the compatibility because we do not have the chip, however, we are planning to do the necessary tests in the near future.
If you would like these tests to be speeded up, you can become a [sponsor](https://github.com/sponsors/ElectronicCats) or [contribute](https://github.com/ElectronicCats/electroniccats-cla/blob/main/electroniccats-contribution-manual.md) to this project, everything is appreciated 😸

wilson-elechouse, 2024-06-14:
We could ship you sample boards of PN7160 and PN7161, if you need them

jazhe, 2024-06-17:
Awesome! 
@wilson-elechouse please email us to support@electroniccats.com to start the process, we appreciate it a lot!