ElectronicCats/ElectronicCats-PN7150 #29: Can't compile to Portenta board

wolfmanus, 2022-02-21:
The "exit status 1" stops the translation. I think the GPIO pins (Portenta is use the GPIO_# format) name/numbering may be the problem.

wolfmanus, 2022-02-25:
It does not translate with correct gpio (arduino #) numbering either.

sabas1080, 2022-02-25:
Can you give more details of the problem? code line

wolfmanus, 2022-02-25:
I wanted to translate (to Portenta H7 M7 core) the original (Electronic Cats PN7150 v1.5.1) example (DetectTags, DetectingReaders).
I found the numbering for the Portenta BreakoutBoard GPIO pins:
`#define PN7150_IRQ   (53) //BreakoutBoard GPIO_3`
`#define PN7150_VEN   (67) //BreakoutBoard GPIO_4`
The Arduino IDE (v1.8.19) can't compile the example:

> C:\Users\wolf\Documents\Arduino\libraries\Electronic_Cats_PN7150\src\Electroniccats_PN7150.cpp: In member function 'uint8_t Electroniccats_PN7150::writeData(uint8_t*, uint32_t) const':
C:\Users\wolf\Documents\Arduino\libraries\Electronic_Cats_PN7150\src\Electroniccats_PN7150.cpp:81:58: error: call of overloaded 'write(uint8_t*&, uint32_t&)' is ambiguous
     nmbrBytesWritten = Wire.write(txBuffer, txBufferLevel);
                                                          ^
In file included from C:\Users\wolf\Documents\Arduino\libraries\Electronic_Cats_PN7150\src\Electroniccats_PN7150.h:29:0,
                 from C:\Users\wolf\Documents\Arduino\libraries\Electronic_Cats_PN7150\src\Electroniccats_PN7150.cpp:18:
C:\Users\wolf\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\2.7.2\libraries\Wire/Wire.h:63:20: note: candidate: virtual size_t arduino::MbedI2C::write(const uint8_t*, int)
     virtual size_t write(const uint8_t* data, int len);
                    ^~~~~
In file included from C:\Users\wolf\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\2.7.2\cores\arduino/api/ArduinoAPI.h:31:0,
                 from C:\Users\wolf\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\2.7.2\cores\arduino/Arduino.h:27,
                 from C:\Users\wolf\Documents\Arduino\libraries\Electronic_Cats_PN7150\src\Electroniccats_PN7150.h:21,
                 from C:\Users\wolf\Documents\Arduino\libraries\Electronic_Cats_PN7150\src\Electroniccats_PN7150.cpp:18:
C:\Users\wolf\AppData\Local\Arduino15\packages\arduino\hardware\mbed_portenta\2.7.2\cores\arduino/api/Print.h:54:20: note: candidate: virtual size_t arduino::Print::write(const uint8_t*, size_t)
     virtual size_t write(const uint8_t *buffer, size_t size);
                    ^~~~~
exit status 1

jazhe, 2022-02-25:
Hello. We apologize for the inconvenience caused,
Unfortunately, this library is not compatible with Portenta boards due to core STM32H747 is not supported yet. 
However, please stay tuned for our updates.

jazhe, 2022-02-25:
Hello again @wolfmanus,
We have updated the library, it is on the branch _fixMbedError_, please try the update on your Portenta board.
We wait for your feedback!
(If you can not download the library, [here you can do it](https://github.com/ElectronicCats/ElectronicCats-PN7150/archive/refs/heads/fixMbedError.zip))

wolfmanus, 2022-02-27:
Thanks, the example can compile well to portenta. I try it on monday to portenta board.
A few days ago i tried with ESP32 board, recognized 1 out of 3 NFC tag (my samsung s21 phone read well all of them), can"t read the serial number nor...

wolfmanus, 2022-03-01:
Hello again,
I tried it with MKR ZERO (the Portenta is hell itself, there is chaos with pin numbering). Recognized 3 out of 5 NFC tag, my samsung s21 phone read well all of them:
### not recognized TAGs:
by my S21 phone, NFC Tools Pro app:
> Tag type: ISO 14443-3A, Innovision R&T Jewel - Topaz 512
Technologies available: NfcA, Ndef
Serial number: XX:XX:XX:XX
ATQA: 0x0C00
SAK: 0x00
Header ROM: 0x124C
Memory information: 512 bytes
Data format: NFC Forum Type 1

by MIKROE PN7150 mudul, v1.5.1 ISO14443-3A_read_block sample
> 15:26:50.543 -> Read ISO14443-3A(T2T) data block 5 with PN7150
15:26:50.543 -> Initializing...
15:26:50.615 -> Waiting for an ISO14443-3A Card ...
15:26:57.216 ->  - Found a card, but it is not ISO14443-3A(T2T)!
15:26:58.816 -> CARD REMOVED!
15:26:58.816 -> Re-initializing...

by MIKROE PN7150 mudul, v1.5.1 DetectTags sample
> 15:31:27.743 -> Detect NFC tags with PN7150
15:31:27.743 -> Initializing...
15:31:27.803 -> Waiting for an Card ...
15:31:30.964 ->  - POLL MODE: Remote activated tag type: 1
15:31:30.964 -> 	SENS_RES = 0x00 0x0C 
15:31:30.964 -> 	NFCID = 
15:31:32.604 -> CARD REMOVED!
15:31:32.604 -> Re-initializing...

Arthulisher, 2022-03-01:
Hello wolfmanus, Could you open a new issue about your new problem, please? in order to provide you with the best support.

closed 2022-06-17 by sabas1080