ElectronicCats/ElectronicCats-PN7150 #32: Configure Setting Failed

anqurarora, 2022-04-30:
I am trying to run the example "ISO 14443-3A_read_blocl" and it is getting stuck at "Configure settings failed!". Upon checking in the source file I found that it is stuck at the "#if NXP_CORE_CONF_EXTN" section of the function (line# 1268). I am using Arduino Uno Board. Any Suggestions to rectify this error?

Thank You.
[image]

IvanAmg, 2022-05-02:
Hello anqurarora!
Would you mind sharing with me if you're using any NFC module with your Arduino Uno Board?

anqurarora, 2022-05-03:
Hi,

Thanks for your reply. I am using Mikroe NFC 2 click.
https://www.mikroe.com/nfc-2-click.

IvanAmg, 2022-05-03:
Thanks for sharing!
Taking a look at it, I've realized that this module works with a  3.3 logic voltage level.
Before trying any software troubleshooting, please make sure to have a proper level shifting circuit.
Have a wonderful day!

anqurarora, 2022-05-03:
I just checked on their website and it does say that it works with both 3.3 and 5V circuits. Which makes sense, because it's getting past the "nfc.connectNCI()" function. Anything else you want me to try? I really want this to work.

Thank you in Advance.

anqurarora, 2022-05-03:
I have connected the uno board with this click board as shown in the attached figure.

[image]

IvanAmg, 2022-05-03:
Oh, I see!
I mentioned the level shifter circuit because that's what the manufacturer recommends.
"This Click board™ is designed to be operated only with a 3.3V logic voltage level, while 5V is used as transmitter input supply voltage of PN7150. A proper logic voltage level conversion should be performed before the Click board™ is used with MCUs with different logic levels."
Anyway, would you mind sharing with me your code? this way I can test it with one of our boards.
Thank you for all the information!

anqurarora, 2022-05-03:
Oh, I'll try to use the level shifter and update you on that. I have used the example code-named "ISO14443-3A_read_block".

Let me know if you need any further information.

Thank You.

blacktorch, 2022-06-16:
I am using Mikroe NFC 2 click as well please which pins are the IRQ and VEN

IvanAmg, 2022-06-17:
Hi @blacktorch!
The "Mikroe NFC 2" is not manufactured by Electronic Cats.
We recommend you to contact its official manufacturer in order to get the help that you want.
Apologies for the inconvenience!

anqurarora, 2022-06-17:
You should be able to find the pin out on their website. Hope it helps.

IvanAmg, 2022-06-21:
> Oh, I'll try to use the level shifter and update you on that. I have used the example code-named "ISO14443-3A_read_block".
> 
> Let me know if you need any further information.
> 
> Thank You.

Hi @anqurarora!
Is there an update on your issue?
Please, let me know!

anqurarora, 2022-06-22:
Hi,

I needed up using different NFC module. Thanks for the help tho.

Thank you.

IvanAmg, 2022-06-22:
Glad to help! 
I'll close this issue now.
Thanks!

closed 2022-06-22 by jazhe