ElectronicCats/ElectronicCats-PN7150 #37: ElectronicCats

DominusDRR, 2022-08-30:
Hi.

When sending the CORE_RESET_CMD, three response bytes are obtained, which are:

0x40, 0x00 and 0x03

In your code simply return an error when "waking up" the NFC

`(void)writeData(NCICoreReset, 4);
    getMessage(15);
    NbBytes = rxMessageLength;
    if ((NbBytes == 0) || (rxBuffer[0] != 0x40) || (rxBuffer[1] != 0x00))
    {
        return ERROR;
    }`

If that happens, what does it mean? What can be done to solve the problem?

I have consulted the NCI specification and there is no information about the values for these 3 bytes.

Eric286, 2022-08-30:
Dear @DominusDRR !

Thank you for contacting us.

Byte 0x00 would be STATUS_OK

Byte 0x03 would be STATUS_FAILED

Byte 0x40 is in other words more special than the other two since it does not only refer to whether the status is good or not.  It would be MFC_Authenticate_REQ (DH asks NFCC to perform MFC authenticate)

"If that happens, what does it mean? What can be done to solve the problem?"

Here what it means is that when you get the message if it does see a byte with 0x40 or 0x00, which would be, the bytes that you get when the status is correct, it will write the data to the board.
But nevertheless, if it does not find any of these two, and finds that the message contains nothing, it would mean that the response was 0x03, which only occurs when the Status fails.

Possible errors to this can be, a wrong key, a time-out triggered during authentication, you are performing two same commands before this, or no message found to write.

Have a nice day!
Kind regards!
Electronic Cats Support Team.

DominusDRR, 2022-08-30:
Hi. Thanks for answering

One last question, the 0x40, in which part of the NCI specification does it fit?

[image]

According to your answer, if I have not misunderstood, the second byte would be Status (0x00), while the third byte (0x03) would be the NCI version:

If so, there is no match with CORE_RESET_RSP
from Table 5.

[image]

DominusDRR, 2022-08-30:
Hi,

I think that 0x40 is the header of the frame, it indicates a response to a command sent from the DH (microcontroller)

[image]

Eric286, 2022-08-30:
Hello @DominusDRR!

Just one thing where did you get the pictures from? (In order to better understand the situation.)

0x40, what it does is ask the NFCC to make an MFC authentication command, with this, name MFC_Authenticate_REQ, to later obtain the MFC Authenticate command status with this name MFC_Authenticate_RSP

So to speak the byte works like REQuest and ReSPonse (REQ and ReSP)

[image]

Have a nice day!
Kind regards!
Electronic Cats Support Team.

DominusDRR, 2022-08-30:
The images were taken from the document called NFC Controller Interface (NCI) Specification and PN7150 User Manual (UM10936)

And it looks like I'm right, the 0x40 has nothing to do with MFC_Authenticate_RSP yet.

The frames have an initial header where it is indicated that they are, for example for CORE_RESET_CMD, according to your own example, 0x20 is sent, which corresponds to

[image]

[image]

And if the NFC responds to the command, the header is 0x40

[image]

Pag 15:

[https://www.nxp.com/docs/en/user-guide/UM10936.pdf](url)

salmg, 2022-08-31:
Hello @DominusDRR 

Yes, you are partially right. The 0x40, it is a response message. It is a combination of MT values + PBF(Packet Boundary Flag) + GID (Group Identifier)

0x00 is 1 bit RFU + 1 bit RFU + 6 bits for Opcode identifier(OID) which basically indicates the level of the previous executed command. In this case 0x00 which means that the command executed is level 0, the first level of commands that could be executed in the PN*. Other type of commands on the same level are CORE_RESET_RSP or CORE_RESET_NTF, for example. You can find the completed list in the Table 102 in the NFC Controller Interface (NCI) Specification from the NFC Forum version 1.0. 

0x03 is the payload length.
[image]

Answering the main question, the core reset helps to write specific core configuration when the PN initializes, establishing a previous configuration. If it fails, means that the previous configuration using the ...CORE_CONF command was not applied correctly and that could be for many different factors.

DominusDRR, 2022-08-31:
Thanks for the explanation. So in the payload (which is 3 bytes) there would be the Status, NCI Version and Configuration Status:

Regards

closed 2022-08-31 by DominusDRR