ElectronicCats/ElectronicCats-PN7150 #38: Which NCI command does the NCIReadTS frame correspond to?

DominusDRR, 2022-09-02:
Hi

I'm following your code to learn the operation of the PN7150.

I've gotten to the part where you want to determine the timestamp stored in NFC Controller

The frame that is sent is the following:

`uint8_t NCIReadTS[] = {0x20, 0x03, 0x03, 0x01, 0xA0, 0x14};`

Because of the OID which is the second byte that is equal to 0x03, you might think that the command corresponds to RF_DISCOVER_CMD

[image]

Am I right?

If I'm wrong, which NCI specification command would it correspond to?

Regards

Eric286, 2022-09-02:
> 
Hello @DominusDRR !

"Am I right?"

Yes, you are right.

Once you send the discovery command (RF_DISCOVER_CMD) and if PN7150 responds with STATUS_OK, then the device will be in the RFST_DISCOVERY state executing the polling loop continuously.

Is this related to your previous issue?

Best Regards!

Support Team

DominusDRR, 2022-09-02:
Thanks for answering, it is not related to the previous question.

I already understand how the frames that travel between the DH and NFCC are specified in NCI.

Regards

Eric286, 2022-09-05:
Hello @DominusDRR 

Can we close the issue?

Best regards!
Support Team.

DominusDRR, 2022-09-05:
I already closed it. I saw that @sabas1080  reopened it and I thought I was going to comment something additional

DominusDRR, 2022-09-06:
Hi.

I'm sorry to open the case again, I'm wrong in my statement, just like @Eric286 

The frame that is sent:

`uint8_t NCIReadTS[] = {0x20, 0x03, 0x03, 0x01, 0xA0, 0x14};`

It doesn't correspond to RF_DISCOVER_CMD as I initially stated.

Corresponds to CORE_GET_CONFIG_CMD, which allows obtaining various parameters from the NFC device.

[image]

The fourth byte, which is equal to 1, indicates the number of parameters and the parameter is 0xA0, 0x14

This parameter corresponds to the EEPROM memory region called DH_EEPROM_AREA_2

[image]

In this region it seems to me that you keep what you call __TIMESTAMP__ and I'm not sure exactly what value it is.

(There is another open thread regarding that topic)

Eric286, 2022-09-06:
Hello @DominusDRR !

I already discussed it with my colleagues and this question will be answered in the corresponding section to be able to close this one.
Which would be issue number 39.

[39](https://github.com/ElectronicCats/ElectronicCats-PN7150/issues/39)

Have a good day!

closed 2022-09-06 by Eric286