ElectronicCats/ElectronicCats-PN7150 #39: Where is __TIMESTAMP__ defined or what is its value?

DominusDRR, 2022-09-05:
Hello.

I'm reviewing your project to learn about the PN7150

I have reached the line where the constant __TIMESTAMP__ appears in Electroniccats_PN7150.cpp

`#if (NXP_CORE_CONF_EXTN | NXP_CLK_CONF | NXP_TVDD_CONF | NXP_RF_CONF)
    uint8_t currentTS[32] = __TIMESTAMP__;`

but I can't find what value it has or where it is defined.

Please, could you tell me where it is?

Or is it something that I should define?

Regards

DominusDRR, 2022-09-05:
Hi.

I answer to myself.

I understand that when the following was done:  

[image]

TIMESTAMP, is obtained, which is an array of 32 bytes

[image]

This array is not saved, it was intended to be done somehow, but the code is commented out:

[image]

However, below, in the code it is used:

[image]

Although I have understood something, I still don't know if I should save and use that array. And if so, I don't understand in code because you don't save it for later use.

DominusDRR, 2022-09-06:
I keep answering to myself.

What I understand so far, if I'm not wrong, is that the 32 bytes of the EEPROM memory are read and don't contain any default information, you must proceed to carry out the steps that correspond to NXP_CORE_CONF_EXTN, NXP_CLK_CONF NXP_TVDD_CONF and finally, in NXP_RF_CONF is saved to a value so that the next time the device is powered up, those parameters are not reconfigured.

That's the idea, and it's commented out as indicated by what's highlighted in yellow in the image below.

[image]

Currently, it seems that this information is read from the EEPROM, but it is discarded as it is not used, and it is not important since any value is simply saved.

So I think I could save any value in that region, to avoid setting the parameters every time the PN7150 is powered up.

The disadvantage of the current code is that it will always be storing meaningless information in that region, thus shortening the life of the EEPROM.

xpeqex, 2022-09-06:
Hi @DominusDRR 
Thank you for contacting us!

__TIMESTAMP__ is a preprocessor macro that expands to current time (at compile time) in the form Ddd Mmm Date hh::mm::ss yyyy, where the time is in 24 hour time, Ddd is the abbreviated day, Mmm is the abbreviated month, Date is the current day of the month (1-31), and yyyy is the four digit year. For example "Wed Jan 18 22:29:12 2012". The __TIMESTAMP__ macro can be used to provide information about the particular moment a binary was built.

This variable is compared with the timestamp of the micro resulting in two possible cases:

1. Both match: the PN7150 performers a normal reset
2. Do not match: PN7150 performs a hard reset needing to be reconfigured.

I recommend you to read the original library provided by NXP, so you can find more information about the library.

Lastly, you can test the library on your own by commenting on these lines and checking if the library works properly after that.

Kind regards,

DominusDRR, 2022-09-06:
Thank you very much for answering about what is TIMESTAMP.

I still have a doubt. This comparison you mention, Does it happen on the PN7150? 

Because in the microcontroller code, I don't see anything that does such a check. Simply note that information is saved to that region of memory whenever the device is initialized.

Regards

wero1414, 2022-09-06:
@DominusDRR this library was based on the library provided from NXP but not all the functions were ported, you can find that library [here](https://www.nxp.com/products/rfid-nfc/nfc-hf/nfc-readers/high-performance-nfc-controller-with-integrated-firmware-for-smart-devices:PN7150), here you can see how it is used in the file NxpNci.c

[image]

Hope this helps you to understand better that variable behaviour and now existence in our library.

DominusDRR, 2022-09-06:
Thank you very much for sharing the information, regarding the last question, I'm answering myself here:

[https://github.com/ElectronicCats/ElectronicCats-PN7150/issues/39#issuecomment-1238413762](url)

(The saving of __TIMESTAMP__ is not implemented, but in the code that is commented, there is the key to do it.

I close the case and thank you very much again.

closed 2022-09-06 by DominusDRR