ElectronicCats/ElectronicCats-PN7150 #41: For the example to read an ISO14443A block, shouldn't the CLIF_ANA_RX_REG = RF_CLIF_CFG_BR_106_I_RXA_P ?

DominusDRR, 2022-09-15:
Hi.

I'm analyzing and following step by step the exampleISO14443A  (read bloack).

I've made it to the top of the WaitForDiscoveryNotification function, where a notification of type RF_INTF_ACTIVATED_NTF is expected.

But I always get a notification of type CORE_GENERIC_ERROR_NTF in which the parameter that reports the error is DISCOVERY_TARGET_ACTIVATION_FAILED.

I thought it was a problem with the antenna, but using an oscilloscope I have determined that the signal is present without problems and without attenuation.

Searching the NXp forum for information about this error, I have found one that indicates that this notification is due to a bad assignment of the parameter called CLIF_ANA_RX_REG.

In the code it is defined as 0xA0, 0x0D, 0x04, 0x34, 0x44, 0x22, 0x00.

[image]

But if the AN11755 document is consulted, for an ISO14443 tag (Type A or Type B) the value must be A0 0D 04 34 44 21 00 or A0 0D 04 46 44 26 00

[image]

Could it be the cause of my problem?

Regards

IvanAmg, 2022-09-15:
Hi! @DominusDRR 
Yes, this actually may cause some errors.
Please, go ahead and let me know if the issue persists!

DominusDRR, 2022-09-15:
Hello. Thanks for answering.

I have changed to the value indicated in document AN11755 ( 0xA0, 0x0D, 0x04, 0x34, 0x44, **_0x21_**, 0x00), but the problem persists

DominusDRR, 2022-09-17:
I answer to myself.

I think I have got something, firstly after carefully checking the antenna and reading the information in document AN11755, in section 3.6.2 it is mentioned about the connection of RX terminals.

In my case, the dimensions of the antenna are 15.5 mm wide by 27.51 mm, that is, an area of 426.41 square mm.

For such a surface, the document recommendation indicates that the RX terminals should be connected directly to the antenna.

[image]

So I made this correction to the antenna (removing Rrx and Crx and directly connecting the RX terminals to the antenna)

[image]

In this way, the error was no longer displayed immediately after Start Discovery.

But the same problem occurred when I presented the tag on the antenna.

I vaguely understood that possibly my error is corrected by modifying the CLIF_ANA_RX_REG registry (section 5.5.3.1).

After doing some tests, I decided to modify the register for a gain of 0 dB and a corner frequency of 250 kHz.

[image]

That is to say that I modified RF_CLIF_CFG_BR_106_I_RXA_P as follows:

//0xA0, 0x0D, 0x04, 0x34, 0x44, 0x22, 0x00,/* RF_CLIF_CFG_BR_106_I_RXA_P   CLIF_ANA_RX_REG */
**0xA0, 0x0D, 0x04, 0x34, 0x44, 0x2C, 0x00,/* RF_CLIF_CFG_BR_106_I_RXA_P   CLIF_ANA_RX_REG */**

And when doing some tests with the tag, I finally got a different notification (I haven't analyzed it in depth yet to determine what it means)

0x61 0x05 0x14 0x01 0x80 0x80 0x00 0xFF 0x01 0x09 0x04 0x00 0x04 0x70 0xB8 0xD1 0x48 0x01 0x08 0x00 0x00 0x00 0x00**

Now, at some point the problem returned to me only once, I imagine that due to some external disturbance this error could occur.

I have another question, when this problem occurs, according to the NCI specification it indicates that the NFC enters a state called RFST_W4_HOST_SELECT in which the 13.56 MHZ signal is present in the antenna (I have verified it, that after sending that notification, in NFC it remains with the aforementioned signal unchanged).

My question is:

Is there a way to determine if the NFC is in that state to resend Start Discovery command?

Regards.

DominusDRR, 2022-09-20:
Hi.

I still have problems, but it seems that I have achieved several things.

I have been able to read the ID of the tag (which is in the frame that I shared) I have also been able to read the data block that the tag has, I was able to detect when it is removed and read it again, it works fine, but the only problem I have is when for some reason detect frame
 
60 07 01 a1

When this notification happens (which happens very rarely), the problem is that it is always happening.

 
That is to say that once I have read it, the IRQ terminal is again set to logical 1, and if said notification is not read, the NFC doesn't respond to other commands.

 
So I remain in an infinite loop always reading that notification.

 
In the infinite loop I wait a while checking that IRQ doesn't return to logical 1, but before reaching the time (500ms), IRQ returns to 1 to indicate the same notification.

Regards

DominusDRR, 2022-09-26:
So far it works pretty well, it's just that problem that very rarely happens.

When the notification (60 07 01 a1) happens several times and does not stop, I have decided to restart (by hardware) the PN7150 and configure everything and everything works again.

Thanks for the support.

[https://www.youtube.com/watch?v=NSJtPNthBEs&lc=UgyZzkabGzK9smoHQN94AaABAg](url)

IvanAmg, 2022-09-28:
Apologies for the late reply.
We'll take your feedback and small experiment for future troubleshooting.
Thank you so much for letting us know about your whole experience while testing our library!