ElectronicCats/ElectronicCats-PN7150 #48: NDEF record reading?

howlingwolven, 2023-02-18:
The documentation and examples are unclear; is there a method that'll return a whole NDEF record stored on a tag? If so, how do I implement it?

Eric286, 2023-02-24:
Hello @howlingwolven!

I am very sorry for the inconveniences that the documentation and examples could cause you by not being clear.

At the moment we do not have NDEF implemented and in fact we have a pull request where you can see how to start implementing it.

https://github.com/ElectronicCats/ElectronicCats-PN7150/pull/25/files

Here I leave the direct link so you can see this and if you want to start using it, however there are still things to do, we are working on it.
Also, if you wish and it is not urgent, you can wait a while for all this to be ready.

Best Regard!
Support Team!

Eric286, 2023-03-06:
Hello @howlingwolven!

I will close the issue.

Please do not hesitate to reach out again if you have any further questions or concerns in the future. 

We are always here to help.

Best regards,

closed 2023-03-06 by Eric286

sabas1080, 2023-07-29:
#25

sabas1080, 2023-09-21:
now available