ElectronicCats/ElectronicCats-PN7150 #53: NDEF write issue in T4T emulation mode

euttsov, 2023-09-06:
Hi guys. I faced the issue here that the offset could be equal to 0. As stated in T4T spec, NLEN field (offset 0) could be updated after NDEF file has been written, but current code does not take it into account.

https://github.com/ElectronicCats/ElectronicCats-PN7150/blob/5c593d13d773e8a040f79621341b9c6754aa3333/src/T4T_NDEF_emu.cpp#L124

Me personally ended up in something like this:

```
    else if (!memcmp(pCmd, T4T_NDEF_EMU_Write, sizeof(T4T_NDEF_EMU_Write)))
    {
        if (eT4T_NDEF_EMU_State == NDEF_Selected)
        {

            unsigned short offset = (pCmd[2] << 8) + pCmd[3];
            unsigned char length = pCmd[4];
			
	    if (offset == 0 && length == 2)
	    {
		    T4T_NdefMessage_size = (pCmd[5] << 8) + pCmd[6];
		    eStatus = true;
	    }
	    else if (offset == 2) 
	    {
		    if (offset + length <= sizeof(T4T_NdefMessageWritten))
		    {
			    memcpy(&T4T_NdefMessageWritten[offset - 2], &pCmd[5], length);
			    pT4T_NdefMessage = T4T_NdefMessageWritten;
			    T4T_NdefMessage_size = length; //(pCmd[5] << 8) + pCmd[6];
		    
			    *pRsp_size = 0;
			    eStatus = true;
		    }
	    }
        }
    }
```

Eric286, 2023-09-11:
Hello @euttsov 

Thank you for sharing your proposal.
Our team will carefully examine your suggestion, this might take a little time, as we want to ensure that any changes do not cause additional problems and are compatible with the overall functioning of the program.

If you have any other suggestions or comments, don't hesitate to share them.

Have a nice day!

Support Team!

euttsov, 2023-09-17:
@Eric286 Hi Eric. Have you been able to check the proposed change? Can I close the issue?

Eric286, 2023-09-19:
Hello @euttsov 

Thank you for your patience. 

Our team has noted your proposed change, and we will be reviewing it in the coming months. We want to ensure that any modifications align with the library's overall functionality and quality standards.

Until then, If you like we can close the issue.

Thank you for your understanding.

Best regards,
Support Team.

closed 2023-09-19 by euttsov