ElectronicCats/ElectronicCats-PN7150 #57: Is this library compatible with PN7120?

AravinthPanch, 2023-10-22:
First of all, thank you very much for the wonderful work in creating this open-source project.

For a small prototype project, I'm using a board with PN7120 and ESP32. 

I'm able to discover the PN7120 via I2C scanning. But I'm not able to communicate with the chip using the example(DetectTags.ino). 

Is this library compatible with PN7120? If yes, doesn't need any specific flags to be set?

***Thanks a lot for the support in advance***

Eric286, 2023-10-23:
Hello @AravinthPanch 

Thank you for your interest in the ElectronicCats PN7150 library.

The ElectronicCats PN7150 library is intended to be used with the PN7150 NFC chip. We have not tested the library with the PN7120, and we cannot guarantee that it will work.

Thank you for your understanding.

Have a nice day!

Support Team.

sabas1080, 2023-10-23:
maybe you can make the library work by making some modifications, currently we do not have a PN7120, all contributions to improve the library are welcome.

AravinthPanch, 2023-10-23:
Thanks, @Eric286 and @sabas1080, for the prompt clarification and suggestion. 

I was confused because topic #PN7210 was tagged in the repo description and there were some declarations for PN7210 in the code.

closed 2023-10-23 by AravinthPanch