ElectronicCats/ElectronicCats-PN7150 #60: Card randomly removed

JVKran, 2024-04-25:
**Used hardware**
An [ESP32-S3-DevKitC-1](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html) connected to the [PN7150 module from ElecHouse](https://www.elechouse.com/product/pn7150_v1/).

**Describe the bug**
At 09:34AM I started the program with a tag directly on top of the scanner. At 10:09AM, the card randomly was removed, according to the library, while it wasn't in reality. From that point onward, it started doing it way more often. See attached logs below.

[serial_monitor.txt](https://github.com/ElectronicCats/ElectronicCats-PN7150/files/15109165/serial_monitor.txt)

**To Reproduce**
PN7150 connected to ESP32-S3. Just running the unaltered DetectTags example. No extra hardware connected. Start the application and place tag in field. Leave it be and it will start doing this without user-intervention. Be patient.

**Expected behavior**
I would expect the CARD_REMOVED event to only fire when I remove a card.

xpeqex, 2024-04-30:
Hi @JVKran 

Thank you for bringing this to our attention and sharing details about the setup. Could you please share with us the following:

- What IDE and version are you using?
- What version of the library?
- What version of the board core?
- Do you have another PN7150 module you can test with? If so, we kindly request you test with this second module.
- Is it possible that any external interference can affect the correct functionality? Some possible causes are:
  - External noise might be affecting the PN7150's communication with the tag.
  - A loose connection or malfunctioning component on the PN7150 module could be causing misreadings.

We are looking forward to yor reply!

xpeqex, 2024-05-08:
@JVKran 

We will close this issue now, however, please feel free to reach out to us again in case you have further suggestions, encounter an issue, or you would like to contribute.

We eagerly anticipate witnessing the remarkable creations you will bring forth! Visit[ hackster.io/ElectronicCats](http://www.hackster.io/ElectronicCats), where you can see others' projects and share yours!

closed 2024-05-08 by xpeqex