ElectronicCats/ElectronicCats-PN7150 #61: Is it compatible with the PN7160?

amin-mm-dev, 2024-10-12:
I was just wondering if this library could be used with the PN7160 chip which is a newer version of the PN7150 and is EMVc compliant.
Not sure if I have to edit any registers or addresses to correctly access and use the Tag Read functionality of the PN7160 chip.

Here's the link: [https://www.nxp.com/products/rfid-nfc/nfc-hf/nfc-readers/nfc-plug-and-play-controller-with-integrated-firmware-and-nci-interface:PN7160](url)

xpeqex, 2024-10-14:
Hello @amin-mm-dev 

Currently, we are working to add support to this chip (no ETA). We will update the documentation and issues once the chip is supported.

Feel free to share your work by [contributing](https://github.com/ElectronicCats/ElectronicCats-PN7150?tab=readme-ov-file#how-to-contribute-) to the repo, so this way the support for the chip can be added.

Alexander96f, 2024-12-31:
Hi everyone,

Are there any updates on adding support for the PN7161 chip? I’m planning to use it with an ESP32 and was wondering if this library is compatible or if work is ongoing to support it.

Thanks!

sabas1080, 2025-01-14:
Hello @Alexander96f we now have a beta version for PN7160 if you would like to test or contribute https://github.com/ElectronicCats/ElectronicCats-PN7150/releases/tag/v3.0.0-beta

Alexander96f, 2025-01-14:
Thanks for the answer!
I am currently waiting for my PCB order, if everything works out I will test the library soon and try to contribute.