ElectronicCats/ElectronicCats-PN7150 #65: PN7161 on ESP32 (3.3V Input) Fails to Generate HF Signal – Seeking Guidance

Alexander96f, 2025-01-24:
Hi everyone,

I’m currently working with the PN7161 NFC controller on a custom board based on the OM27160A1 design. The setup uses an ESP32 microcontroller with 3.3V input voltage at the chip. However, I’m encountering an issue where no HF signal is being generated at TX1/TX2.

Here’s what I’ve done so far:

    I’m using the v3.0.0-beta and running the example code (attached below).
    My connections are as follows:
        PN7160_IRQ → GPIO 4
        PN7160_VEN → GPIO 5
        I²C SDA → GPIO 21
        I²C SCL → GPIO 22
    Despite loading the code successfully, the serial console only outputs:

```
    Restarting...
    Waiting for a Card...

```
    Using a multimeter, I checked TX1 and TX2 pins, but I see no HF activity (unlike my previous work with the MFRC522, where I could measure the 13.56 MHz signal).

I suspect this could be related to the 3.3V power supply of the ESP32 and the issue described in the following discussions:
[
    PN7160 TXLDO stays at 0V if VBAT is connected to 3V3](https://community.nxp.com/t5/NFC/PN7160-TXLDO-stays-at-0V-if-VBAT-is-connected-to-3V3/m-p/1484040)
    [PN7160 works but PN7161 doesn't with a similar design](https://community.nxp.com/t5/NFC/PN7160-design-works-but-PN7161-doesn-t-work/m-p/1566504#M9882)

Here’s what I’ve tried so far:

    Rechecked the schematic and wiring, and I can’t identify any obvious issues.
    Modified some configuration settings in the library, but I’m not sure I fully understand how to adapt it to my setup.

My goal is to perform basic UID reading of common MIFARE RFID cards. Is there something I’m missing in the hardware design, configuration, or initialization process? Could the problem be related to the ESP32 providing 3.3V input to the PN7161?

Any suggestions, ideas, or experiences with a similar setup would be greatly appreciated. Thanks!

Here’s the example code I’m using:
**#include "Electroniccats_PN7150.h"
#include <Wire.h> // For I²C communication

#define PN7160_IRQ 4   // Pin for IRQ
#define PN7160_VEN 5   // Pin for VEN
#define PN7160_ADDR 0x2A // Default I²C address

Electroniccats_PN7150 nfc(PN7160_IRQ, PN7160_VEN, PN7160_ADDR, PN7160);

// Function prototypes
String getHexRepresentation(const byte* data, const uint32_t numBytes);
void displayCardInfo();

void setup() {
  Serial.begin(115200);
  while (!Serial)
    ;
  Serial.println("Detect NFC tags with PN7160");

  // Initialize I²C
  Wire.begin(21, 22);  // SDA = GPIO 21, SCL = GPIO 22
  Serial.println("Initializing...");

  if (nfc.connectNCI()) {  // Wake up the board
    Serial.println("Error while setting up the mode, check connections!");
    while (1)
      ;
  }

  if (nfc.configureSettings()) {
    Serial.println("The Configure Settings is failed!");
    while (1)
      ;
  }

  // Read/Write mode as default
  if (nfc.configMode()) {  // Set up the configuration mode
    Serial.println("The Configure Mode is failed!!");
    while (1)
      ;
  }
  nfc.startDiscovery();  // NCI Discovery mode
  Serial.println("Waiting for a Card...");
}

void loop() {
  if (nfc.isTagDetected()) {
    displayCardInfo();

    if (nfc.remoteDevice.hasMoreTags()) {
      nfc.activateNextTagDiscovery();
      Serial.println("Multiple cards are detected!");
    }

    Serial.println("Remove the Card");
    nfc.waitForTagRemoval();
    Serial.println("Card removed!");
  }

  Serial.println("Restarting...");
  nfc.reset();
  Serial.println("Waiting for a Card...");
  delay(500);
}

// Helper functions remain unchanged
String getHexRepresentation(const byte* data, const uint32_t numBytes) {
  String hexString;

  if (numBytes == 0) {
    hexString = "null";
  }

  for (uint32_t szPos = 0; szPos < numBytes; szPos++) {
    hexString += "0x";
    if (data[szPos] <= 0xF)
      hexString += "0";
    hexString += String(data[szPos] & 0xFF, HEX);
    if ((numBytes > 1) && (szPos != numBytes - 1)) {
      hexString += " ";
    }
  }
  return hexString;
}

void displayCardInfo() {
  char tmp[16];

  while (true) {
    switch (nfc.remoteDevice.getProtocol()) {
      case nfc.protocol.T1T:
      case nfc.protocol.T2T:
      case nfc.protocol.T3T:
      case nfc.protocol.ISODEP:
        Serial.print(" - POLL MODE: Remote activated tag type: ");
        Serial.println(nfc.remoteDevice.getProtocol());
        break;
      case nfc.protocol.ISO15693:
        Serial.println(" - POLL MODE: Remote ISO15693 card activated");
        break;
      case nfc.protocol.MIFARE:
        Serial.println(" - POLL MODE: Remote MIFARE card activated");
        break;
      default:
        Serial.println(" - POLL MODE: Undetermined target");
        return;
    }

    switch (nfc.remoteDevice.getModeTech()) {
      case (nfc.tech.PASSIVE_NFCA):
        Serial.println("\tTechnology: NFC-A");
        Serial.print("\tSENS RES = ");
        Serial.println(getHexRepresentation(nfc.remoteDevice.getSensRes(), nfc.remoteDevice.getSensResLen()));

        Serial.print("\tNFC ID = ");
        Serial.println(getHexRepresentation(nfc.remoteDevice.getNFCID(), nfc.remoteDevice.getNFCIDLen()));

        Serial.print("\tSEL RES = ");
        Serial.println(getHexRepresentation(nfc.remoteDevice.getSelRes(), nfc.remoteDevice.getSelResLen()));

        break;

      case (nfc.tech.PASSIVE_NFCB):
        Serial.println("\tTechnology: NFC-B");
        Serial.print("\tSENS RES = ");
        Serial.println(getHexRepresentation(nfc.remoteDevice.getSensRes(), nfc.remoteDevice.getSensResLen()));

        Serial.println("\tAttrib RES = ");
        Serial.println(getHexRepresentation(nfc.remoteDevice.getAttribRes(), nfc.remoteDevice.getAttribResLen()));

        break;

      case (nfc.tech.PASSIVE_NFCF):
        Serial.println("\tTechnology: NFC-F");
        Serial.print("\tSENS RES = ");
        Serial.println(getHexRepresentation(nfc.remoteDevice.getSensRes(), nfc.remoteDevice.getSensResLen()));

        Serial.print("\tBitrate = ");
        Serial.println((nfc.remoteDevice.getBitRate() == 1) ? "212" : "424");

        break;

      case (nfc.tech.PASSIVE_NFCV):
        Serial.println("\tTechnology: NFC-V");
        Serial.print("\tID = ");
        Serial.println(getHexRepresentation(nfc.remoteDevice.getID(), sizeof(nfc.remoteDevice.getID())));

        Serial.print("\tAFI = ");
        Serial.println(nfc.remoteDevice.getAFI());

        Serial.print("\tDSF ID = ");
        Serial.println(nfc.remoteDevice.getDSFID(), HEX);
        break;

      default:
        break;
    }

    if (nfc.remoteDevice.hasMoreTags()) {
      Serial.println("Multiple cards are detected!");
      if (!nfc.activateNextTagDiscovery()) {
        break;
      }
    } else {
      break;
    }
  }
}
**
Any help or advice is greatly appreciated!

Alexander96f, 2025-01-25:
Ok I dont really know what I did but I got it working now and can read Uids from tags with my custom pcb with pn7161 and esp32 c6

To get the PN7161 working on my device, I made the following changes:
	1.	In the .h file, set NXP_TVDD_CONF to 1 instead of 2.
	2.	In the .cpp file, changed line 738 to:

if (NxpNci_CONF_size == 0) {
instead of 
if (NxpNci_CONF_size != 0) {

I dont know how this line worked for you before because I couldnt find how NCI_CONF_SIZE is getting declared to something other than 0 in the code?

So the config wasnt even loaded to my PN7161 all the time. Also I changed something on the  NxpNci_TVDD_CONF_3rdGen for my device (but I dont know yet if this was also necessary for 3,3V will test again later). 

Maybe this can help you to improve it, thank your for your work!

SjonHert, 2025-08-06:
Hi, I have a nearly identical setup, with the OM27160A1 devkit for the PN7161 and an ESP32-WROOM-32D (though I'm hoping to port it to stm32g070rb soon if it works. I run into the same problem being unable to detect an RF signal with my oscilloscope, nor get a response when holding a tag nearby. It stays in the loop 

```
Restarting... 
Waiting for a Card...
```

I have made the changes you proposed to set NXP_TVDD_CONF to 1 and the if statement of NxpNci_CONF_SIZE an == instead of !=, but that does not solve it for me.

I have tried with both the code you posted above, as with the example code DetectTags.ino (changing only the pins for IRQ and VEN and assigning chip model to PN7160).

``` 
#define PN7160_IRQ (18)
#define PN7160_VEN (19)
#define PN7160_ADDR (0x28)

Electroniccats_PN7150 nfc(PN7160_IRQ, PN7160_VEN, PN7160_ADDR, PN7160); // fourth additional argument for chip model PN7150 or PN7160 
```

Would you be willing to share your source code in which you got it working (including the driver tweaks you made) so I can test if it works for me?

Either way I see this issue is still open, so it hasn't been resolved yet? I subscribed to stay up to date.

MProx, 2025-08-08:
Following because I have the same issue. Custom PCB with a schematic very similar to OM27160A1. The device responds to I2C communication correctly but I can't see any evidence that an RF field is being generated. I've also tried the code modifications noted by @Alexander96f.