ElectronicCats/ElectronicCats-PN7150 #66: Tag Emulation on ESP32 and PN7160 – iPhone Doesn’t Show NFC Notification

SmartSystemMZ, 2025-04-09:
Hi everyone,
I’m attempting to emulate an NFC card on an ESP32 WROOM using an OM27160A1HN (PN7160) demo board. 
The end goal is to have iPhone users (and others) simply tap their phone on the device and see a notification (banner) that redirects them to a URL in Safari—no additional apps required.

What works so far:
-  The phone is indeed recognized by the PN7160 (the library’s isReaderDetected() returns true).
-  I can manually scan the tag with the NXP TagInfo app, and it does show the URL—but TagInfo also says “tag not fully readable” and labels many fields as “unknown”.
-  Another app, NFC Tools, only shows minimal information: “Tag Type: ISO 7816, Technology: Unknown…”

What doesn’t work:
- On iPhone (and some Android devices), no normal NFC banner/notification appears at all. Therefore, the user isn’t automatically prompted to open Safari with the URL.

What I’ve tried:
- The library’s NDEFMessage and NDEFRawMessage examples from ElectronicCats-PN7150.
- Attempting to set a single URI record with correct NDEF headers (MB=1, ME=1, SR=1, TNF=1, etc.).
- Checking various prefix bytes (e.g., 0x03 for “https://”).

Observations:
- Both TagInfo and NFC Tools identify the PN7160 device as ISO 7816 but with “unknown” fields, which suggests that iOS might not recognize it as a standard NFC Forum Tag Type 4.
- When scanning real NFC Forum Type 2 or Type 4 tags, the iPhone shows a popup or banner.

To confirm if there’s a complete Type 4 Tag emulation happening, including the capability container (CC File) so that iOS properly shows a notification.

Questions:
Do I need to explicitly configure a CC File or anything else in the PN7160 to make sure it’s recognized as a standard Type 4 NDEF Tag?
Has anyone successfully used the PN7160 with an ESP32 to emulate a URL that triggers iOS’s native NFC banner?
Any troubleshooting tips or code snippets that show a fully working Type 4 Emulation (with correct NDEF records) on the PN7160?
I appreciate any help or insights. I’m hoping to get iOS to show the usual NFC URL notification/bubble without needing a separate app.
Thanks in advance!

xpeqex, 2025-04-09:
Hi @SmartSystemMZ 

I understand that you expected this to work seamlessly with our library. By default, the example NDEFSendMessage emulates an NFC Tag Type 4. It's important to note that this functionality was tested with the PN7150, and support for the PN7160 is still under development. We truly appreciate feedback like yours, as it helps us improve the library.

We will perform tests on our end to try and replicate the issue. If you don’t mind me asking, are you using the library from the [Releases](https://github.com/ElectronicCats/ElectronicCats-PN7150/releases) section, or did you download it directly from the PN7150/60 branch [here](https://github.com/ElectronicCats/ElectronicCats-PN7150/tree/PN7150/60)? If downloading from the releases section, what version?

SmartSystemMZ, 2025-04-09:
Hi Carlos,

First of all, thank you very much for your work on the library – I really appreciate it!

I didn’t necessarily expect everything to work seamlessly out of the box, I was rather unsure whether this kind of setup would work at all. So I’m very grateful for the effort you’re putting into it.

I’m currently testing with the PN7150/60 branch that I downloaded directly from GitHub. Unfortunately, I was never able to get the board to initialize with Beta 3.0.1 – only Beta 3.0.0 worked for me so far.

Thanks again, and I’m looking forward to further updates!

Best regards
Mario

xpeqex, 2025-04-10:
Hey Mario @SmartSystemMZ ,

Thanks for the details. I have already tested this and what I found some interesting behaviors:

- On Android devices, I was able to read NDEF messages and have it automatically open the browser to the defined URL. This with Samsung, Motorola and Huawei phones.
- The NFC Tools and TagInfo (NXP) applications on Android identify the module as an ISO 14443-4 Type 4 tag. NFCTools correctly displays the information.
- On iPhone, the module was not detected, so no notification was received and the web browser was not prompted to open.
- Strangely, a notification was received a few minutes later on one of the iPhones.
- The NFCTools app on the iPhone detects the module as an ISO 7816 tag, as you mentioned.

My setup:
1. ESP WROOM 32 Dev Kit
2. PN7160 NFC I2C module from ELECHOUSE
3. NDEFSendMessage sample code. I have only modified a few definitions to make it work with PN7160.

That said, iOS may not support custom tag emulations and need real, passive NDEF tags. Probably due to the certifications issued by Apple. As for Android devices, do they also identify the PN7160 module as an ISO 7816 tag?

In any case, we will keep looking into this and see if there is an workaround/fix to make this work. Let me know if you need me to send you some video or the code I mention in this message.

Do not hesitate to share with us any other doubt about this.

SmartSystemMZ, 2025-04-11:
Hi Carlos @xpeqex,

Thank you very much for your continued testing!
Since I didn’t have an Android device nearby, I wasn’t able to test it myself. But I borrowed a Samsung, and I can confirm that it’s working on Android  👍

> Strangely, a notification was received a few minutes later on one of the iPhones.

I noticed almost the same behavior: it worked once, but I couldn’t reproduce it. I thought I might have just imagined it.
But after reading your message, I gave it another serious try… and I actually managed to reproduce it!

I held the unlocked iPhone close to the PN7160, then immediately locked it – and suddenly, the notification popped up.
From what I understand, locking the iPhone interrupts the communication.

Unfortunately, I don’t know the library well enough to stop the communication at the correct point after discovery.
So I just tried calling nfc.reset() before sending the message — and to my surprise, it worked!
```
void loop() {
  Serial.print(".");
  if (nfc.isReaderDetected()) {
    Serial.println("\nReader detected!");
    
    Serial.println("Reset Reader");
    nfc.reset();
    
    Serial.println("Sending NDEF message...");
    nfc.sendMessage();
    Serial.print("\nWaiting for an NDEF device");
  }
}
```

Maybe this insight can help you address the issue in the right place in the code.

In any case, thank you again for your input!

Best regards,
Mario

xpeqex, 2025-04-11:
Hey Mario @SmartSystemMZ,

Thank you for the update and for sharing the workaround you found, I can confirm this makes it work on iOS but affects the performance of the code, not a big deal but noticeable.

In any case, this has been raised to our dev team, and they will start working on this.

We will keep you updated, feel free to share any other questions.

xpeqex, 2025-04-16:
@SmartSystemMZ 

Just as an update, the dev team is already working on this, and they will launch a release once it is resolved, thank you for your patience.

xpeqex, 2025-04-25:
Hey @SmartSystemMZ 

We appreciate your patience in this matter, we are still working on it as this issue seems to be more complex than expected. For now, we have detected that this only happens on Apple devices and that the sending functions are not received as expected on these devices due to the way they handle NFC communications, but this does not indicate that it is impossible. 

For the moment, the workaround you mentioned above seems to us to be the best solution for this problem.

> So I just tried calling nfc.reset() before sending the message — and to my surprise, it worked!
>
> ```
> void loop() {
>   Serial.print(".");
>   if (nfc.isReaderDetected()) {
>     Serial.println("\nReader [...]

We will close this, however, we invite you to keep an eye on the new [releases](https://github.com/ElectronicCats/ElectronicCats-PN7150/releases) of the library and its changelog so you can be aware of what's new.

closed 2025-04-25 by xpeqex

SmartSystemMZ, 2025-04-29:
Hi @xpeqex thanks for the update, sadly i wracked my NXP dev board, so i couldn't do any further testing. I now received PN7160 Boards from Elechouse...
Could you recommend any other Chip for this purpose?
Thx Mario