ElectronicCats/ElectronicCats-PN7150 #68: Default NDEF Read Example Doesn't Output Any NDEF Message (Message Always Empty or Buffer Too Small)

Markker98, 2025-06-15:
When using the official Electroniccats_PN7150 example to read NDEF messages from an NTAG215 (Type 2) tag, the tag is detected successfully, and the interface/protocol are correctly reported (e.g., FRAME / T2T / PASSIVE NFC A). However, readNdefMessage() does not trigger the callback or populate the NDEF message — even when the tag contains a valid NDEF payload like "hello" (confirmed using NXP TagInfo and NFC Tools).

To Reproduce
Steps to reproduce the behavior:

Flash the ReadNDEFMessage example to a supported MCU (e.g., ESP32).

Write a short NDEF text message (e.g., "hello") to a NTAG215 using NFC Tools or NXP TagWriter.

Tap the tag.

Observe: the tag is detected, but no message is printed. Callback is not triggered.

Expected behavior
The callback (setReadMsgCallback()) should be called and output the NDEF message contents.

Observed behavior
Only this is printed:

Initializing...
Waiting for a Card...
Interface: FRAME
Protocol: T2T
Technology: PASSIVE NFC A
Remove the Card
Card removed!
Restarting...
Waiting for a Card...

xpeqex, 2025-06-16:
Hi @Markker98 

Thanks for letting us know this. I will try to replicate this on my end, just for reference, are you using a PN7150 or PN7160?

xpeqex, 2025-06-24:
We will close this issue now, if you want to continue with it, please feel free to reopen it. In cany case, we will keep this in mind for future reference.

closed 2025-06-24 by xpeqex