ElectronicCats/ElectronicCats-PN7150 #69: Cannot set custom UID in card emulation mode on PN7160 (always random UID)

asepwildan, 2025-07-16:
Hi! 👋

First of all, thank you for this great library.

I’m using ESP32 + PN7160.
I’m trying to use card emulation mode with a custom UID (CSN), but when I scan the emulated card using a reader, it always shows a random UID, not the one I set.

`uint8_t customUID[] = {0x08, 0x48, 0xAC, 0x31};`
`bool uidResult = nfc.configureSettings(customUID, 4);`
`Serial.print("configureSettings() returned: ");`
`Serial.println(uidResult);`

`Target UID: 0848AC31`
`configureSettings() returned: 1`
 `configureSettings() reports ERROR`

`Waiting for NFC reader...`
`✅ NDEF message sent successfully!`
`E (15500) i2c.master: I2C hardware NACK detected`
`...`
`Ready for next scan...`

I expected the card emulation would use the UID I defined (08 48 AC 31) and the reader would detect this UID but current behavior the reader always receives a random UID, and configureSettings() reports an error. 

my question is, Is it actually possible to set a fixed UID in card emulation mode on PN7160?
Or is this a hardware limitation (for example, UID is always generated internally by the PN7160 chip)?

AndreaZGuz, 2025-07-16:
Hi @asepwildan!

This appears to be a limitation of the module.
Since the PN7160 does not have its own UID and functions primarily as an NFC reader rather than a tag, it is not possible to change the identifier of this module. The UID is a property of NFC cards/tags.
It would be possible to emulate a tag with a specific UID using the PN7160 (although this is not a native feature), but this is also limited by the capabilities of the microcontroller being used.

AndreaZGuz, 2025-07-21:
Hello @asepwildan!

We will close this issue now, however, please feel free to reach out to us again in case you have further suggestions, encounter an issue, or would like to contribute.

I wish you a fantastic day ahead!
Best regards,

closed 2025-07-21 by AndreaZGuz